belyalov/tinyweb #1: Remove dot prefix from mime_types data structure

0x7162, 2018-01-04:
Minor: I suppose dot prefix could be removed from data structure, i.e.
...
                  'jpg': 'image/jpeg',
...
ext = fname[idx+1:]
...

https://github.com/belyalov/tinyweb/blob/4ea409cf004b6b684b452af486cbc8f4bd77ee80/tinyweb/static.py#L9-L19

belyalov, 2018-01-05:
Could you please describe what are the benefits / issues you're trying to add/resolve?
The most popular way how to extract file ext is:
```python
>>> import os
>>> filename, file_extension = os.path.splitext('/path/to/somefile.ext')
>>> filename
'/path/to/somefile'
>>> file_extension
'.ext'
```
From [stackoverflow](https://stackoverflow.com/questions/541390/extracting-extension-from-filename-in-python/).

0x7162, 2018-01-09:
The only benefit here is readability based on fact you are not using this extension to access any file.
It is really minor issue - you can drop it easily :-)

belyalov, 2018-01-22:
Could you please update your PR with unittests / solve conflicts to get this issue resolved?

belyalov, 2018-03-01:
Closing due to no activity, feel free to reopen.

closed 2018-03-01 by belyalov