belyalov/tinyweb #18: ImportError: no module named 'tinyweb'

Kortenbach, 2018-12-19:
Make sure you are running the latest version of TinyWeb before reporting an issue.
Fresh download today

**Device and/or platform:**
ESP8266

**Description of problem:**
After flashing .bin firmware file:
import network --> ok
import tinyweb (copy-pasted name, no typo...) --> ImportError: no module named 'tinyweb'

**Expected:**
Successfull import

**Traceback (if applicable):**
```bash

```

**Additional info:**

belyalov, 2018-12-20:
Thanks for reporting, looks to be `travis` build bug.
ETA for fix is tomorrow

closed 2018-12-20 by belyalov

belyalov, 2018-12-20:
With [release-1.3.2 image](https://github.com/belyalov/tinyweb/releases/download/v1.3.2/firmware_esp8266-v1.3.2.bin) it works well:
```python
>>>
>>> import network
>>> import tinyweb
>>>
>>> tinyweb.
__class__       __name__        __path__        webserver
server
```

Let me know if it still does not work for you.

Kortenbach, 2018-12-20:
Thank you so much!
I will report back to you

commit a4d8367b (2018-12-20): Update travis build script, fixes #18. Update micropython version for release image build.