belyalov/tinyweb #20: chunk mode send json

hyzeer, 2019-07-26:
Hi , thank you very much for your wonderful framework , but I regret to find an error in processing NON-ASCII characters by using the framework .

```
for chunk in res:
     await resp.send('{:x}\r\n'.format(len(chunk)))
     await resp.send(chunk)
     await resp.send('\r\n')
```
the above codes in server.py at line 337-340 , "len(chunk)" refer to the numbers of the character variable "chunk" , not the bytes length .  It's OK with the return of the function "len( )" , when the character variable is ASCII type , but for the NON-ASCII type , such as a Chinese character , the character number is 1 in function "len( )" , and the length is 3 bytes . So , this function will cause a loss in transmission .

belyalov, 2019-07-26:
Hi @hyzeer,

thanks for reporting this issue - I never tried to send non ASCII chars through JSON, so you're totally right! :)

It actually easy to fix by
```python
            chunk_len = len(s.encode('utf-8'))
            await resp.send('{:x}\r\n'.format(chunk_len))
```

The question here is - do you guys always using `utf8` as default encoding?
In Russia we do use `utf8` as default, so I wonder to we need to make encoding configurable or we can just hardcode it - in order to save bytecode / RAM.

hyzeer, 2019-07-26:
Hi @belyalov , thanks for your replay.
Yes, most of us do using utf-8 as default encoding anytime. :)

Could you release new firmware after fixed? I can do it, but I not cross compile environment. Thanks!

closed 2019-07-28 by belyalov

belyalov, 2019-07-28:
Fix just landed and [new firmware](https://github.com/belyalov/tinyweb/releases/tag/v1.3.4) is ready.

Enjoy!

commit 0b76fc1b (2019-07-28): Properly handle UTF strings restful_resource_handler()
Fixes #20