belyalov/tinyweb #21: UTF-8 char garbled in json response by frontend

hyzeer, 2019-08-02:
Hi, thanks for your new firmware,  it can handle UTF-8 char properly,  but it has new problem about garbled. my codes is the following:

```
@websvr.resource('/config/wifi/aplist', method='GET')
async def connect_to_ap(data):
    gc.collect()
    sta = network.WLAN(network.STA_IF)
    sta.active(True)
    ap_list = sta.scan()
    resp = json.dumps({
        "msg": "OK",
        "result": [
            {"ssid": x[0].decode(), "authmode": x[4], "rssi": x[3]}
            for x in ap_list
        ]
    })
    for s in resp:
        yield s
```

the above codes works correctly if values of resp['result'] contains no UTF-8 chars, otherwise the frontend receiving will be garbled, like this:
[image]

I guess this due to Content-Type filed not is 'application/json;charset=utf-8' in Response-Header, but I can't debug it..

belyalov, 2019-08-05:
Hi,

I think we can add override for encoding, let me try it (however it will take time, busy days)

hyzeer, 2019-08-06:
Thanks for your reply in busy days.  :)

another point, at v1.3.3, chars length handle error, I did this to avoiding the error:

```
 for x in ap_list:
      difference = len(x[0].decode().encode('utf-8')) - len(x[0].decode())
      yield '"ssid": "{}",'.format(x[0].decode()) + 'a' * difference

```

I used the above codes, utf-8 chars can be handled propely by tinyweb and decoded propely by the browser. I don't know why it happened. 🤣