belyalov/tinyweb #22: LmacRxblk:1 error

2niuhe, 2019-12-04:
Make sure you are running the latest version of TinyWeb before reporting an issue.

**Device and/or platform:**
ESP8266
micropython latest version

**Description of problem:**
when my phone lost reconnect to the AP,
LmacRxblk:1 occur.
**Expected:**

**Traceback (if applicable):**
```bash

```

**Additional info:**
I am new to python,maybe you don't close sock when CTRL+C interrupt.
maybe you can check this link
[https://forum.micropython.org/viewtopic.php?t=3602]

belyalov, 2019-12-05:
Don't you mind to share your program?
It is currently unclear...

2niuhe, 2019-12-25:
I have solve the problem.It's caused by another mqtt package.
Thanks for your tinyweb project again.

closed 2019-12-25 by 2niuhe