belyalov/tinyweb #26: No CaptivePortal support?

UnexpectedMaker, 2020-03-01:
I've tried everything, but I can't get TinyWeb to serve me a webpage when my DNS is setup for a CaptivePortal.

DNS is set as a wildcard for 192.168.4.1, and I can surf to that page, and see the website, but it doesn't auto load it on connection to the AP.

It works correctly with other web servers (MicroWebSrv2, Sockets server).

Is there something I am missing? Should this be working, or if not, can this be added?

Thanks,
Seon

UnexpectedMaker, 2020-03-02:
The issue here seems to be no ability to do a catch-all route, like I've done before and like what MicroWebSrv2 has, so I added it myself with the following code:

```
class webserver:
    def __init__(self, request_timeout=3, max_concurrency=3, backlog=16, debug=False, handlenotfound=False):
        """Tiny Web Server class.
        Keyword arguments:
            request_timeout - Time for client to send complete request
                              after that connection will be closed.
            max_concurrency - How many connections can be processed concurrently.
                              It is very important to limit this number because of
                              memory constrain.
                              Default value depends on platform
            backlog         - Parameter to socket.listen() function. Defines size of
                              pending to be accepted connections queue.
                              Must be greater than max_concurrency
            debug           - Whether send exception info (text + backtrace)
                              to client together with HTTP 500 or not.
            handlenotfound  - If True, then if 'find_handler' can't find a matching url, and '/' is in 
                              the dictionary, it will use that instead. If False, it behaves like previous
                              functionality and 'find_hanlder' returns (None,None)
        """
```
and then in _find_handler

```
def _find_url_handler(self, req):
        """Helper to find URL handler.
        Returns tuple of (function, opts, param) or (None, None) if not found.
        """
        # First try - lookup in explicit (non parameterized URLs)
        if req.path in self.explicit_url_map:
            return self.explicit_url_map[req.path]
        # Second try - strip last path segment and lookup in another map
        idx = req.path.rfind(b'/') + 1
        path2 = req.path[:idx]
        if len(path2) > 0 and path2 in self.parameterized_url_map:
            # Save parameter into request
            req._param = req.path[idx:].decode()
            return self.parameterized_url_map[path2]
        # No handler found
        if self.handlenotfound and '/'.encode() in self.explicit_url_map:
            return self.explicit_url_map['/'.encode()]
        else:
            return (None, None)
```

It's not a fantastic solution, so maybe something better can be added to support a "not found" fallback route?

Cheers :)

belyalov, 2020-03-03:
Hey! :)

I remember I was able to make captive portal work using `tinyweb` + `tinydns`. Since it was awhile back ago and I've switched from python away (to STM32), a few points that may help you:
- setup dns: https://github.com/belyalov/iot-platform/blob/master/devices/open_neopixel_controller/app.py#L132
- setup captive portal: https://github.com/belyalov/iot-platform/blob/master/devices/open_neopixel_controller/app.py#L91
- captive portal itself: https://github.com/belyalov/iot-platform/blob/master/platform/utils/captiveportal.py
- tinydns: https://github.com/belyalov/tinydns

belyalov, 2020-03-03:
So basically solution (apart from DNS part) is to just handle few URLs, like:
```python
    # Add URLs
    web.add_route('/generate_204', captive_portal)
    web.add_route('/hotspot-detect.html', captive_portal)
    web.add_route('/library/test/success.html', captive_portal)
```

UnexpectedMaker, 2020-03-03:
Ohai!
I didn't know you had `tinydns` - Excellent.
Yeah, my issue with adding all of the "usual captive portal routes" for the different platforms is that I have no way of testing them other than Mac/iOS based ones, so I was wanting to just do the "catch-all" approach.

Catch all works for me because for this part of the application I can always be assured all traffic is for the captive portal experience. 

I have to say though, I've been struggling to work out how to get TinyWeb routes working non my own classes (I'm fairly new to MP and Python in general) - but your `open_neopixel_controller` just solved that problem for me... it's quite clear from that how to do it now, so BIG THANKS!

belyalov, 2020-03-03:
🎉 
Cool, you're welcome :)

belyalov, 2020-03-04:
Seems you're happy with solution, closing issue - feel free to reopen it :)

closed 2020-03-04 by belyalov

UnexpectedMaker, 2020-03-16:
Hi, I'm back on this after a long hiatus - Sorry for the delay.

I just can't get tinydns to work - or I just don't understand what I am doing (most likely).

I'm a asyncio noob, so that doesn't help at all.

I have all of the captive routes setup, I still have my own tinyweb hack always falls back to '/' if it can't find a route, and no matter what I do, I can't get it to serve a page once I connect to my AP.

The exact same code using MicroDNSSrv works as expected.

I'd gladly keep using MicroDNSSrv, but I still have this outstanding issue:
https://github.com/belyalov/tinyweb/issues/25

I've tried setting the DNS to 0.0.0.0:53 and to the IP of the AP, but same issue. I'm thinking maybe the DNS is just not running as I can't see how to make it "run forever" like I can with tinyweb. 

I've followed your implementation examples from your int-platform example, but I just can't get it to work :(

So I really don't know where to go from here.