belyalov/tinyweb #27: Still can't get captive portals working :(

UnexpectedMaker, 2020-03-16:
Hi, I'm back on this after a long hiatus - Sorry for the delay.

I can't seem to be able to re-open this previous issue so I have created a new one:
https://github.com/belyalov/tinyweb/issues/26

I just can't get tinydns to work - or I just don't understand what I am doing (most likely).

I'm an asyncio noob as well, so that doesn't help at all.

I have all of the captive routes setup, I still have my own tinyweb hack always falls back to '/' if it can't find a route, and no matter what I do, I can't get it to serve a page once I connect to my AP.

The exact same code using MicroDNSSrv works as expected.

I'd gladly keep using MicroDNSSrv, but I still have this outstanding issue:
#25

I've tried setting the DNS to 0.0.0.0:53 and to the IP of the AP, but same issue. I'm thinking maybe the DNS is just not running as I can't see how to make it "run forever" like I can with tinyweb.

I've followed your implementation examples from your int-platform example, but I just can't get it to work :(

So I really don't know where to go from here. I'd love to only be using tinyweb & tinydns for my MP web projects.

UnexpectedMaker, 2020-03-18:
I'm moving this to tinydns, as it's clearly not tinyweb related ;)

closed 2020-03-18 by UnexpectedMaker

belyalov, 2020-03-18:
Hey,

As I remember there was some issue with binding to `0.0.0.0` - maybe related WiFi interface readiness (e.g. you need to wait until DHCP done / re-bind once interface configured).

I can try to get repro, within next 2 weeks, but if it is too late - you may want to play around network configuration..

Thanks for feedback! :)