belyalov/tinyweb #29: Updated tinyweb for new uasyncio?

UnexpectedMaker, 2020-03-26:
Damien has merged the new uasyncio into mainline for ESP32 (and other ports) so I was hoping that we could get a new "compatible" version of tinyweb that uses it.

From what I understand (been told by Jim Mussared) tinyweb is using some internal asyncio APIs that wont carry over to the newer uasyncio. I'm not sure how hard it will be to move it, but the plan is to have the new uasyncio ship in 1.13 (frozen), so things are going to likely break when that happens.

I'm happy to be a bleeding edge tester on this as I'm keeping the project I am working on the latest mainline right now, so I'm happy to break things on my side to help get it done.

Cheers!
Seon

belyalov, 2020-03-27:
Hey, thanks for heads up!

I'll try to update it this weekend.. )

UnexpectedMaker, 2020-03-27:
Wicked, thanks @belyalov !

mvincm, 2020-03-27:
Hello.

I'm here to test it just push the beta version :-)

Best regards 
MvincM

UnexpectedMaker, 2020-04-03:
Hey @belyalov, any progress on looking into this? No pressure, just curious.
I see more uasyncio improvements have got into master, so it's get some love from Damien atm, so would be a good time to get any tinyweb/uasyncio issues addressed if needed. 
Cheers,
Seon

belyalov, 2020-04-05:
Hey,

No progress so far - but I'm going to give it a try today.. :)

belyalov, 2020-04-06:
Made some progress.
It seems working well, however, I hit few issues:
- `unix` ports does not have `uasyncio` enabled by default, so unable to run unittests at this time.
- new `uasyncio` does not currently have `serve_forever()` so some workarounds needed from app side

Anyway it is now looks better and really close to python3.x:
```python
import tinyweb
import uasyncio

# Create web server application
app = tinyweb.webserver(host='0.0.0.0', port=8081)

# Index page
@app.route('/')
async def index(request, response):
    # Start HTTP response with content-type text/html
    await response.start_html()
    # Send actual HTML page
    await response.send('<html><body><h1>Hello, world! (<a href="/table">table</a>)</h1></html>\n')

async def main():
    await app.start()

if __name__ == '__main__':
    uasyncio.run(main())
    uasyncio.get_event_loop().run_forever()
```

Hopefully will get back in few days.. :)
-----
Micropython issue: [extmod/uasyncio: add serve_forever()](https://github.com/micropython/micropython/issues/5870)
WIP is on [new_uasync_io](https://github.com/belyalov/tinyweb/tree/new_uasync_io) branch.

UnexpectedMaker, 2020-04-06:
Excellent progress, thanks @belyalov !

c0d3z3r0, 2020-04-09:
reader.readexactly() isn't implemented, yet

UnexpectedMaker, 2020-05-30:
Hey @belyalov - I guess this isn't going to happen? I've been in a bit of a holding pattern waiting for this, as I need to move to a newer MP version for other fixes/features, and without this being compatible with the new asyncio, I'll have to remove it and use microdot instead, which would suck as I have the features using tinyweb fully implemented in my project already.

bsecker, 2020-07-07:
@belyalov are there any plans to support the new uasyncio? Much like @UnexpectedMaker my v1.12 firmware has asyncio v3.0 baked in, and I would love to be able to use tinyweb with it.

Thanks for the awesome library :-)

PR #36 (2020-12-05): uasyncio v3 fix: loop.call_soon -> loop.create_task
Missed a function that's not in uasyncio v3

`loop. call_soon` is only uasyncio v2, but `loop.create_task` works for all versions

closes #29
closed 2020-12-05 by belyalov