belyalov/tinyweb #32: Question: tinyweb on esp8266 compared to other frameworks

flokain, 2020-11-28:
Hi there,

Thanks for building tinyweb!

I am working on an enduser-friendly, lowcost, DYI IOT device that will run on esp8266 and at the moment it uses tinyweb for providing a REST-API and a WebUI. I need to be very thrifty with memory utilization. It might seem like an odd question, but would you recommend tinyweb over other servers like [picoweb](https://github.com/pfalcon/picoweb) or [MicroWebSrv](https://github.com/jczic/MicroWebSrv) (not sure if this would run on esp8266 properly), and why?

If so, I would consider contributing to the project once i feel comfortable with results in my own [project](https://github.com/flokain/Smart-Cube).

Thank you!

metachris, 2020-12-03:
I like tinyweb because it's small and efficient, and works with the default MicroPython uasyncio library. Note that you need to apply #33 for MicroPython 1.13 (uasyncio v3) compatibility. 

picoweb is not compatible with standard MicroPython uasyncio, but requires the pycopy asyncio library. So you need to use either this or run pycopy instead of stock MicroPython.

flokain, 2020-12-03:
Hi chris, Thanks for your thoughts!

belyalov, 2020-12-04:
Hi @flokain,

Initially I was implemented it just to make it small and well tests in order to be used in my personal projects.. )
Regarding the differences - I'm not that much aware of them, I can say about tinyweb - it is small and efficient, as @metachris already mentioned :)

closed 2021-02-11 by flokain