belyalov/tinyweb #58: Logger calls exc instead of exception

sjefferson99, 2024-04-14:
Unless I have missed something obvious, the webserver includes the standard micropython logging module and then attempts to call log.exc() instead of log.exception()

I have manually patched my server code to adjust this and exceptions are now correctly being surfaced in the logs instead of an error that log does not have function exc.

There are 3 occurrences in the _handler function lines 479, 484 and 488.

sjefferson99, 2024-04-14:
This hasn't actually fixed it, I was premature in assuming the new errors were my exceptions and not further issues with the logging module. The old exc function took the argument e which is not the same as msg that exception expects. My error handling knowledge is not great and I guess this is where I dig in and make it better.

The logger module changes behaviour in this [commit](https://github.com/micropython/micropython-lib/commit/d717b04cb32f6badb7de10667b0b9525c59ae087).

The notes do not seem to give a rationale for removing exc or how to modify code that relied on it, so will need to work out what function to call (presumably exception) and how. I'll post back here if I get that far.

sjefferson99, 2024-04-14:
Looks like the best approach is to use log.exception(f"<Text detailing specific conditions>. Original error {e}"). This is working well for me now surfacing the error in my resource code.

gbsallery, 2024-05-11:
Have just hit this issue as well. Any problems with merging @sjefferson99's patch?

closed 2024-05-14 by belyalov